rohanrhu/gdb-frontend #2: How to run with a Qt project?

vadi2, 2020-02-15:
I've got a Qt project that I can build and run either with qmake or cmake - how can I use it in this frontend?

I've read through the instructions, but it's not clear on how to get started.

rohanrhu, 2020-02-15:
It must be like typical GDB debugging. Load executable, add a breakpoint and run. Also you can use it with `gdbserver` for remote debugging.

There is an article for GDB/Qt debugging:
https://community.kde.org/Guidelines_and_HOWTOs/Debugging/Debugging_with_GDB#Improving_your_gdb_experience_for_KDE.2FQt_applications

vadi2, 2020-02-15:
My typical is different from your typical! I simply use an IDE - Qt Creator or CLion - which manages it all for me.

I got it to work by doing `file <path to binary>`, `set cwd <working directory the binary should run under>`, and `run`.

closed 2020-02-15 by vadi2

rohanrhu, 2020-02-15:
Notice: When you load an executable with file browser, it will set cwd to executable directory.

vadi2, 2020-02-15:
Yeah. In my case I need it to be a different one, because a couple of files are loaded relatively to the binary.