rohanrhu/gdb-frontend #3: How to load additional source files?

vadi2, 2020-02-15:
I'm a little confused by the file manager - when I load and run my binary, it loads _some_ of the source files that it uses, but not all of them. If I try to load a source file manually, I get the following stacktrace (it doesn't seem to be copyable as text):

[image]

Observe how so few files under `src/` are available. If I try and load TLuaInterpreter.cpp, I get the following stacktrace.

My project is https://github.com/mudlet/mudlet if you'd like to replicate it. How can I view all source files?

rohanrhu, 2020-02-15:
Look this: https://github.com/rohanrhu/gdb-frontend#gdb-related-issues-and-tips

vadi2, 2020-02-15:
Fair enough... but then it means I have to step through from the start of the program to where I want to every time. Can I use this like I use gdb elsewhere - have the source files open, set a breakpoint, and when it's hit, the program is paused?

rohanrhu, 2020-02-15:
It is necessary for one time. GDB will remember sources on re-loading executables.

I will implement open additional source file feature for pre-setting breakpoints before executables.

vadi2, 2020-02-16:
That would be super helpful. As I have a GUI application, I can't really step through the files until I get to where I need to go - it enters the main Qt loop, and that is a lot of files to step through then.

rohanrhu, 2020-04-05:
https://github.com/rohanrhu/gdb-frontend/issues/8#issuecomment-609462481

closed 2020-04-05 by rohanrhu