rohanrhu/gdb-frontend #5: WebSocket Problems?

VegaDeftwing, 2020-02-17:
[image]

Not sure what the issue is really. I tried seeing if for some reason turning off firewalld and disabling all my adblockers and other extensions. Failing that, I tried opening it in Chromium, with the same result.

Firefox gives a similar error "Firefox can’t establish a connection to the server at ws://127.0.0.1:5552/." with the issue again being in the main .js file on line 509. , so it's clearly browser agnostic and other local services like Netdata run fine.

rohanrhu, 2020-02-17:
Can you provide `./gdbfrontend` output and output for `tmux a -t gdb-frontend`?

VegaDeftwing, 2020-02-17:
[image]

even starting ./gdbfrontend with -V doesn't yield any more useful output really

rohanrhu, 2020-02-17:
What is your `tmux` version and which OS are you using?

jibanes, 2020-02-17:
I have the same issue on debian 9 running tmux 2.3

VegaDeftwing, 2020-02-17:
Arch Linux 5.5.4 , tmux 3.0_a-1

rohanrhu, 2020-02-17:
Try updating tmux to 2.8 at least or empty `tmux.conf` content. I will add a check tmux version check for fallback

rohanrhu, 2020-02-17:
> Arch Linux 5.5.4 , tmux 3.0_a-1

Very interesting.. clean your `tmux.conf` until the solution.

VegaDeftwing, 2020-02-17:
> > Arch Linux 5.5.4 , tmux 3.0_a-1
> 
> Very interesting.. clean your `tmux.conf` until the solution.

I'm no longer at that desktop, I was only able to confirm those version numbers because my laptop is on Arch too. I'll let you know if cleaning the tmux.conf fixes things when I get home in ~3 hours.

jibanes, 2020-02-17:
I have no tmux.conf , same problem.

rohanrhu, 2020-02-17:
> I have no ~/.tmux.conf , same problem.

It has its own `tmux.conf`: https://github.com/rohanrhu/gdb-frontend/blob/master/tmux.conf

Clean its content but don't remove the file.

jibanes, 2020-02-17:
I cleared tmux.conf
I get popups "Connection closed to GDBFrontend server!" on 5551
gdb is "running" in 127.0.0.1:5550

VegaDeftwing, 2020-02-17:
Tried it from my laptop, same "Connection Closed Message" even if tmux.conf in the gdb-frontend folder is cleared here as well

VegaDeftwing, 2020-02-17:
The Tmux session does open now:
[image]

rohanrhu, 2020-02-17:
I think carriage returns may occur this problem but im not sure.

```
  Offset: 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 	
00000000: 73 65 74 20 2D 67 20 73 74 61 74 75 73 2D 62 67    set.-g.status-bg
00000010: 20 62 6C 75 65 0D 0A 73 65 74 20 2D 67 20 73 74    .blue..set.-g.st
00000020: 61 74 75 73 2D 66 67 20 77 68 69 74 65 0D 0A 73    atus-fg.white..s
00000030: 65 74 20 2D 67 20 73 74 61 74 75 73 20 6F 66 66    et.-g.status.off
00000040: 0D 0A 73 65 74 20 2D 67 20 6D 6F 75 73 65 20 6F    ..set.-g.mouse.o
00000050: 6E                                                 n
```

Can you try this? https://github.com/rohanrhu/gdb-frontend/commit/6c9723ad6f0ac5897c30197e6e68bec26ea1676e

rohanrhu, 2020-02-17:
> The Tmux session does open now:
> [image]

I have not tested GDB 9 yet. I will look that.

VegaDeftwing, 2020-02-17:
That didn't appear to do anything. Tmux is working fine from what I can tell. Intreguringly, on the laptop I'm getting a bit more output, despite getting the same error in the browser, this is regardless of what's in the tmux.conf file. Also, on the laptop, for no apparent reason, if I go to 127.0.0.1:5551/terminal, the terminal does load briefly now (this is dependent on tmux.conf) but I still get the connection error message and the page gives me no choice but to reload
[image]

VegaDeftwing, 2020-02-17:
> > The Tmux session does open now:
> > [image]
> 
> I have not tested GDB 9 yet. I will look that.

I'd imagine that's our problem. I'll see if I can install 8 and point it to use it. Gimme a bit

rohanrhu, 2020-02-17:
Can you paste outputs for these commands on GDB shell?

`shell echo $CWD`
`python import sys;print(sys.version)`
`python print(dir(server))`

VegaDeftwing, 2020-02-17:
I just tried downgrading to 8.3.1, no change. Putting them back up to 9.
``` shell echo $CWD``` gives no output

``` (gdb) python import sys;print(sys.version)
3.8.1 (default, Jan 22 2020, 06:38:00) 
[GCC 9.2.0]
```

rohanrhu, 2020-02-17:
Did you try python print(dir(server)) on tmux gdb-frontend session?

VegaDeftwing, 2020-02-17:
> Did you try `python print(dir(server))` on tmux gdb-frontend session?

Not above, but I just did and the output is exactly the same

VegaDeftwing, 2020-02-17:
Strangely, in the tmux session I can't even do the print(sys.version), 
[image]

rohanrhu, 2020-02-17:
> Strangely, in the tmux session I can't even do the print(sys.version),
> [image]

`print()` is python function like `python import sys;print(sys.version)`

rohanrhu, 2020-02-17:
> > Did you try `python print(dir(server))` on tmux gdb-frontend session?
> 
> Not above, but I just did and the output is exactly the same

Can you paste these outputs?
`python import os;print(os.getcwd())`
`python import server;print(dir(server))`

VegaDeftwing, 2020-02-17:
> > Strangely, in the tmux session I can't even do the print(sys.version),
> > [image]
> 
> `print()` is python function like `python import sys;print(sys.version)`

ope, my bad. Currently in class so trying to listen and type at the same time. It does print in the tmux session:
[image]

rohanrhu, 2020-02-17:
It seems that file is different from https://github.com/rohanrhu/gdb-frontend/blob/master/server.py

Can you paste this output? `python import server;print(server.__file__)`

VegaDeftwing, 2020-02-17:
Yep, you're right. It's using ```/usr/lib/python3.8/site-packages/server.py```

rohanrhu, 2020-02-17:
> Yep, you're right. It's using `/usr/lib/python3.8/site-packages/server.py`

Can you try this? https://github.com/rohanrhu/gdb-frontend/commit/874e5b5e31694f29f7d812d4fb0d4f1db853f77c

VegaDeftwing, 2020-02-17:
cant test it to really confirm right now, (laptop is about to die) but that does seem to have worked!

rohanrhu, 2020-02-17:
Im sorry for your laptop :/ Did https://github.com/rohanrhu/gdb-frontend/commit/6c9723ad6f0ac5897c30197e6e68bec26ea1676e solve the tmux issue?

VegaDeftwing, 2020-02-17:
> Im sorry for your laptop :/ Did 6c9723a
> solve the tmux issue?

Battery died, so responding from email on phone. Yes, that fixed the tmux
problem.

rohanrhu, 2020-02-20:
Can we close this issue?

VegaDeftwing, 2020-02-20:
Yeah, didn't actually realize it wasn't already. My bad.

closed 2020-02-20 by VegaDeftwing