rohanrhu/gdb-frontend #6: No module named socketserver

AlanFoster, 2020-02-18:
Context: I'm trying to run this frontend with a cross compiled gdb:

```shell
$ /home/z/opt/cross/bin/i686-elf-gdb --version
GNU gdb (GDB) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

I installed gdb with:

https://github.com/AlanFoster/os_concepts/blob/d1c34e5009f459b57afa25df06535251c2bbaa09/install_crosscompiler.sh#L55-L72

After cloning and running ` ./gdbfrontend` with the cross compiled gdb:

```
 ./gdbfrontend --gdb-executable=/home/z/opt/cross/bin/i686-elf-gdb --verbose
Listening on 127.0.0.1: http://127.0.0.1:5551/
|---------------------------------------------------------------------|
| Open this address in web browser: http://127.0.0.1:5551/terminal/   |
|---------------------------------------------------------------------|
2020/02/17 15:45:41 Loading config file at: gotty.conf
2020/02/17 15:45:41 GoTTY is starting with command: tmux a -t gdb-frontend
2020/02/17 15:45:41 Permitting clients to write input to the PTY.
2020/02/17 15:45:41 HTTP server is listening at: http://127.0.0.1:5550/
```

Visiting `http://127.0.0.1:5551/terminal/` within the browser:

[image]

Visiting `http://127.0.0.1:5550` within the browser:

[image]

I've confirmed that running `./gdbfrontend` and using the "native" gdb works fine. The version details for that are:

```
$ gdb --version
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
```

I'm thinking the error is python2/3 related? I can see if I can rebuild gdb with the appropriate python version - let me know your thoughts :+1:

rohanrhu, 2020-02-18:
Is your GDB's embedded python version 2? Can you paste output for `python import sys;print(sys.version)` on GDB shell?

AlanFoster, 2020-02-18:
@rohanrhu Thanks! I wasn't sure how to check the python version :+1:

Cross compiled:

```
$ /home/z/opt/cross/bin/i686-elf-gdb -q -ex "python import sys;print(sys.version)" -ex "quit"
2.7.17 (default, Nov  7 2019, 10:07:09) 
[GCC 7.4.0]
```

Host:

```
gdb -q -ex "python import sys;print(sys.version)" -ex "quit" 
3.6.8 (default, Oct  7 2019, 12:59:55) 
[GCC 8.3.0]
```

I'll try and recompile gdb with the `--with-python` flag and try again

rohanrhu, 2020-02-18:
I'm not sure is python3 required for remote GDB but the GDB that runs gdb-frontend requires embedded python3.

https://docs.python.org/2/library/socketserver.html
This document says:
`Note The SocketServer module has been renamed to socketserver in Python 3. The 2to3 tool will automatically adapt imports when converting your sources to Python 3.`

It must be solved with GDB & embedded python3.

AlanFoster, 2020-02-18:
@rohanrhu I was able to recompile gdb with python by installing:

```
python3 python3-distutils python3-dev
```

And providing the `with-python` flag to `gdb/configure`:

```
../${GDB}/configure --target=$TARGET --prefix="$PREFIX" --disable-nls --enable-languages=c,c++ --without-headers --with-python=/usr/bin/python3
```

Now the python version is as expected:

```
/home/z/opt/cross/bin/i686-elf-gdb -q -ex "python import sys;print(sys.version)" -ex "quit"
3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0]
```

Everything boots up fine now, but attaching to the qemu process gives me error:

[image]

```python
GDBFrontend] gdb_on_stop()                                                                                       
0x0000fff0 in ?? ()                                                                                               
Traceback (most recent call last):                                                                                
  File "/home/z/gdb-frontend/api/debug.py", line 313, in getState                                                 
    block = selected_frame.block()                                                                                
RuntimeError: Cannot locate block for frame.                                                                      
                                                                                                                  
Traceback (most recent call last):                                                                                
  File "/home/z/gdb-frontend/api/debug.py", line 60, in _exec__mT                                                 
    output = callback(*args, **kwargs)                                                                            
  File "/home/z/gdb-frontend/api/debug.py", line 786, in disassembleFrame                                         
    block = frame.block()                                                                                         
RuntimeError: Cannot locate block for frame.                                                                      
                                                                                                                  
Traceback (most recent call last):                                                                                
  File "/home/z/gdb-frontend/api/debug.py", line 313, in getState                                                 
    block = selected_frame.block()                                                                                
RuntimeError: Cannot locate block for frame.                                                                      
                                                                                                                  
Traceback (most recent call last):                                                                                
  File "/home/z/gdb-frontend/api/debug.py", line 60, in _exec__mT                                                 
    output = callback(*args, **kwargs)                                                                            
  File "/home/z/gdb-frontend/api/debug.py", line 786, in disassembleFrame                                         
    block = frame.block()                                                                                         
RuntimeError: Cannot locate block for frame.                                                                      
                                                                                                                  
[GDBFrontend] gdb_on_exited()              
```

I ran qemu with:

```
$ qemu-system-i386 -S -gdb tcp::1234
```

I then tried to connect to `localhost:1234` within `gdb-frontend` and got the above error.

I can use gdb directly, without gdb-frontend, and attaching with `target remote localhost:1234` just fine though. Is there anything I can do to help debug this?

rohanrhu, 2020-02-18:
You can ignore `RuntimeError: Cannot locate block for frame.` errors. I had tested it with my OS on QEMU. They are not an error in fact. GDB just does not know frame bounds of QEMU bios/firmware stuff. Disassembly view disassembles in bounds of current frame for now. I will add new options like offsets for it.

I think you loaded kernel elf. Can you try adding breakpoint on your kernel entry-point and continue?

AlanFoster, 2020-02-18:
@rohanrhu Thanks! It looks like everything runs fine, and the trackback/error log is only a distraction 🎉

rohanrhu, 2020-02-18:
> @rohanrhu Thanks! It looks like everything runs fine, and the trackback/error log is only a distraction 🎉

Nice. Thank you for your feedback.

closed 2020-02-18 by rohanrhu