rohanrhu/gdb-frontend #8: Can't load sources at all

joshhansen, 2020-04-04:
I have tried and tried but nothing in the user interface lets me load sources.

I've loaded a debug executable complied using Rust. Other debuggers find the sources automatically. In this case, no sources load automatically and there is no apparent means of loading sources at all. No widget in the UI allows this. No folders appear. No files appear. No source text appears.

If you can provide a usable debugger for Rust in Linux you will be my heroes forever.

But right now, I sadly cannot use your tool.

rohanrhu, 2020-04-04:
Look this: https://github.com/rohanrhu/gdb-frontend#gdb-related-issues-and-tips

When you step over a line that calls a function from the linked object, GDB will get source files from linked object.

Loading additional sources is a TODO: https://github.com/rohanrhu/gdb-frontend/issues/3

Also, after trying that trick, you can provide more information on especially Rust debugging for this issue.

Until when "additional sources" feature is come, you can open source files on Developer Tools with following command:
```javascript
GDBFrontend.components.gdbFrontend.openSource({file: {path: '/path/to/source.rust'}})
```
This will open your source file on source viewer.

rohanrhu, 2020-04-05:
I added open source button to sources section (also shortcut: Ctrl+O). (https://github.com/rohanrhu/gdb-frontend/commit/0ad35702ed729058ac973da6df956d81adcd5347)

I'm closing this issue but also I will add an another feature like adding additional source folders feature.

joshhansen, 2020-04-05:
I ran the latest version and see the new add sources button. I think something else may be causing my difficulty. I loaded my executable, and added its source file, and then clicked "Run". The terminal gives a stacktrace followed by "gdb.error: The program is not being run." I could not copy and paste the error from the terminal so I pieced it together from the HTML inspector in my browser:

```
GNU gdb (GDB) Fedora 8.3-7.fc30
Copyright (C) 2019 Free Software Foundation, Inc.                       
License GPLv3+: GNU GPL version 3 or later &lt;
This is free software: you are free to change and redistribute it. 
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu". 
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
http://www.gnu.org/software/gdb/bugs/
Find the GDB manual and other documentation resources online at:
http://www.gnu.org/software/gdb/documentation/
For help, type "help". 
Type "apropos word" to search for commands related to "word". 
(gdb) warning: Missing auto-load script at offset 0 in section .debug_gdb_scripts
of file /home/josh/Projects/Umpire/target/debug/umpire-train-ai. 
Use `info auto-load python-scripts [REGEXP]\' to list them.
(gdb) Traceback (most recent call last):
  File "/home/josh/Projects/External/gdb-frontend/api/debug.py", line 60, in _exec__mT
    output = callback(*args, **kwargs)
  File "/home/josh/Projects/External/gdb-frontend/api/debug.py", line 488, in cont
    gdb.execute("c")
gdb.error: The program is not being run.

(gdb) 
```

rohanrhu, 2020-04-05:
It is fixed.

rohanrhu, 2020-04-05:
> I ran the latest version and see the new add sources button. I think something else may be causing my difficulty. I loaded my executable, and added its source file, and then clicked "Run". The [...]

I'm not sure what was this problem exactly. You can tell me about if last revision (https://github.com/rohanrhu/gdb-frontend/commit/124ee0f7bd9683852c52e859666baf69a427005d) solved the problem.

closed 2020-04-05 by rohanrhu

joshhansen, 2020-04-05:
The latest revision has the same error, but I am able to see the loaded source file now.

rohanrhu, 2020-04-05:
> The latest revision has the same error, but I am able to see the loaded source file now.

Are you sure you are clicking to run button not continue button?

Run button has a **bug icon**.

joshhansen, 2020-04-06:
I was clicking the button to the right of the bug icon. The bug button looked like the logo so I think I just ignored it.

rohanrhu, 2020-04-06:
> I was clicking the button to the right of the bug icon. The bug button looked like the logo so I think I just ignored it.

So, there is no any problem now?

joshhansen, 2020-04-06:
It runs now though I am unable to set breakpoints in the UI. You may also want to reconsider the icon---it is common for debugging to start with the triangle "play" icon, e.g. VSCode. It may be confusing to others as well as myself. Or maybe not, I'll leave that to you!

rohanrhu, 2020-04-06:
What is the problem for setting breakpoints? If you are setting breakpoints before loading executable, GDB will disable them when you load an executable. In this case, you can re-enable that breakpoints on "Breakpoints" section on the right side. Otherwise, you must provide more information about that.

joshhansen, 2020-04-06:
When I click the red dot in the left margin to set a breakpoint for a line, nothing happens.

rohanrhu, 2020-04-06:
> When I click the red dot in the left margin to set a breakpoint for a line, nothing happens.

Very interesting.. There maybe an issue about espacially Rust debugging. Can you open a new issue with more information about the issue like rust source (that causes this issue), video, screenshots, and instructions to reproduce the same issue.