rohanrhu/gdb-frontend #9: Unspecified error when using gdbfrontend

pawlos, 2020-06-19:
I've tried to use GDB frontend on Ubuntu running in WSL. I can start it buf if I want to use the UI in the browser I'm getting either 'Path not found' or 'An error occured'.

Cmd to run gdbfrontend:

     gdbfrontend --verbose --workdir=/c/temp/ctf/ropemporium/task-write4

Inside gdbfrontend:
    
    file ./write4
    break main
    r

After that - click on stack frame gives me this:
[image]

Click on any fine in the source on the left side gives this:

[image]

Am I doing something wrong here?

rohanrhu, 2020-06-19:
Is there any output in GDB shell? You can access GDB shell with `http://127.0.0.1:5551/terminal/` or `tmux a -t gdb-frontend`.

pawlos, 2020-06-19:
Not much really (at least nothing that would scream error). 

https://gist.github.com/pawlos/d3e7c66b7bf085cf756669e9a410daf6

rohanrhu, 2020-06-19:
> Not much really (at least nothing that would scream error).
> 
> https://gist.github.com/pawlos/d3e7c66b7bf085cf756669e9a410daf6

It says `Reading symbols from ./write4...(no debugging symbols found)...done.` Is the executable compiled with gcc's `-g` option with WSL gcc?

You can compile and link executables with debug symbols like: `gcc -o executable executable.c -g`

pawlos, 2020-06-19:
Yeah, there's no source as it's not even compiled by me but if source not present I was expecting to see some disassembly. I cannot use gdbfrontend w/o source code to see disassm? Is it only working with source code?

rohanrhu, 2020-06-19:
Yes for now. I will add a disassembly tab/view in addition to disassembly view on the right side. It is a TODO.

For frames, I think there is an issue about no source on switching frames. I will fix it for displaying frame address instead of source file/line and fix switching problem for no source situtiations. When it is fixed we will able to seeing registers for each frame for no source situtiations.

There is no a registers view but you can watch registers with adding registers like `$register` to watches view. (Example: `$rax`)

rohanrhu, 2020-06-21:
I added debugging without debug symbols support and improved something. After testing, I will release a new version. You can try last revision (https://github.com/rohanrhu/gdb-frontend/commit/695acd43e894b0a2d748008abe75c4f85c71f26c) and tell me if it is ok or is there any error.

pawlos, 2020-06-22:
It works :)

Additional observations, in the call stack, after clicking on __libc_start_main() there's an error that cannot find source, but after dismissing it the disassembly shows up correctly - not sure if an error is needed (guess depending on user needs?). In the terminal the following errors are present:

https://gist.github.com/pawlos/7592b0a4fb11dab6d5110105e17d4bd6

Note, it was gdb with pwndbg so there might be some interference. Later I can check without pwndbg.

rohanrhu, 2020-06-22:
Missing source errors dont repeat for same sources no longer.

Does that errors cause a problem as functionality? I will add a behaviour to ignore it.

pawlos, 2020-06-22:
Nope, looks like everything works as expected but I'll test a bit more in the upcoming days.

rohanrhu, 2020-06-22:
Ignored that errors. https://github.com/rohanrhu/gdb-frontend/commit/8a94938902d2818892273e4a62f79ebeda053b93

closed 2020-06-26 by rohanrhu