rohanrhu/gdb-frontend #10: Theme + font configuring?

jstaursky, 2020-09-20:
Any way of editing the theme or the font used? 
Awesome Project BTW.

rohanrhu, 2020-09-21:
Hi, theming is actually simple but we need a CSS file configuration. I will add a setting for choosing themes. Theme will be a CSS file but we also need a theme configuration for the editor so theme folders will contain theme CSS and editor theme.

rohanrhu, 2020-10-01:
I just remembered plugin system. Themes can be written as plugins. You can make themes as plugins with name format of `theme_THEMENAME`.

There is a plugin development tutorial: https://github.com/rohanrhu/gdb-frontend/wiki/Plugin-Development-Tutorial

I added an example theme plugin: https://github.com/rohanrhu/gdb-frontend/tree/master/plugins/theme_light

You can enable it with commenting the line of `"theme_light"`:
https://github.com/rohanrhu/gdb-frontend/blob/c8a9549a621e9c68c754231b5ebdc41fa0323022/config.py#L20-L23

For editor theme, I added a method to `FileTabs` component. You can see that in example theme:
https://github.com/rohanrhu/gdb-frontend/blob/c8a9549a621e9c68c754231b5ebdc41fa0323022/plugins/theme_light/frontend/js/theme_light.js#L12-L14

**Note:** You must restart GDBFrontend when you edit `config.py`.

---

Also I will add an option for changing themes. (Theme plugin names must be start with `theme_`, so there will be a select box for switching between themes.)

rohanrhu, 2020-10-02:
I added theme commands to GDB shell.

So you can change theme like:
```
gf-theme light
```
or for default theme:
```
gf-theme default
```

closed 2020-10-02 by rohanrhu