rohanrhu/gdb-frontend #11: View base class elements

jstaursky, 2020-09-20:
Is there a way to view variables apart of the base class when only a derived class variable is available in scope?

rohanrhu, 2020-09-21:
I did not undertand your question. Can you explain more with screenshots and description?

jstaursky, 2020-09-21:
No problem,
[image]
Notice how the derived variable `doot` has its own window that lists its member `derived_var` but in order to see its base class variable `base_var` an new window instance must be opened.
it would be nice to bring this information into 1 window to remind you that the class you are working with is a derived class. And also to show information about variables in the base class.

rohanrhu, 2020-09-30:
I made base class members visible. You can test it and I will make a new release after testing.

closed 2020-10-02 by rohanrhu

jstaursky, 2020-10-03:
nice seems to work well (haven't tested super thoroughly outside the example). Just a suggestion but it might be useful to have the base members to be put inside a submenu so that the derived members are clear from the base members something like

```text
doot members ...
--------------------------------------------------------------------------------
 <foo> = { ... (expand menu)
--------------------------------------------------------------------------------
```
Or some other way to discern the base class members from the derived class members.
It's not that big of a deal for a small example like this but for bigger projects there can be a lot of both base class members and derived members.

rohanrhu, 2020-10-03:
Actually inherited classes are shown but i excluded them. So i can incldue them again but expanding inherited class feature is not implemented yet but still that can be useful for understanding member hierarchy.