rohanrhu/gdb-frontend #12: "Load File" directory traversal should sort entries alphabetically

ell1e, 2020-10-08:
"Load File" directory traversal should sort entries alphabetically. Right now it seems to be completely random making it kind of unusable:

[image]

rohanrhu, 2020-10-08:
Added a JavaScript sort. (https://github.com/rohanrhu/gdb-frontend/commit/cdb562c7292c684cc31635ee48e1de2197b2f74e)

ell1e, 2020-10-09:
Is it already supposed to show up as sorted with the latest git version? Sadly, it still seems to be mixed up even though I can see that added `.sort()` line when I browse in Inspect mode into FileBrowser.js

rohanrhu, 2020-10-09:
It must be solved with https://github.com/rohanrhu/gdb-frontend/commit/6cc9fff73be12f87317aa0f12bc46a037a7a3a30

ell1e, 2020-10-09:
It works now, very nice work :blush:  :+1:

closed 2020-10-09 by ell1e