rohanrhu/gdb-frontend #13: "Load File" when picking an executable only loads one single source file, and doesn't let me browse the others in the tree

ell1e, 2020-10-08:
This is what happens when I load my executable, as you can see just `main.c` pops up:

[image]

This wouldn't be an issue if the tree shown to the left would actually allow me to browse the project folder for the other files. However, for some reason it doesn't, and opening them all up manually via "Load File" would take forever (since load file is unsorted as mentioned in #12 and doesn't start out in the directory I previously picked something from, making it really tedious to open up multiple files form the same folder). So some sort of "Show other files from this directory" context menu or button in the tree view to the left would really help, or alternatively a fix for whatever causes the other source files to not get detected in the first place. If you need more info for fixing that detection please tell me what you need, I'm happy to provide what I can.

rohanrhu, 2020-10-08:
Hi, you can look this: https://github.com/rohanrhu/gdb-frontend#gdb-related-issues-and-tips

GDB does not give sources of linked object **until stepping a line that calls a function from the linked object once**. You can add break point a line and step it once, then you will see sources from linked object hereafter during the session.

ell1e, 2020-10-08:
Ah I see, that at least explains the technical side. However in my opinion that is however extremely inconvenient to use, since I already know where I want to add my breakpoints so I don't want to manually step around or look up the line number in an external editor to do that. If I wanted to look up my breakpoints manually in an external text editor I could have used command line gdb to start with, the entire point of a frontend for me is to solve such inconveniences.

I suggest you just provide a way to easily expand the file tree to the left to work around this. For example, just giving the folders a context menu or a "fully expand" icon to the right of each folder entry would be useful.

rohanrhu, 2020-10-08:
Usually, a breakpoint and stepping once a random line shows all sources. But i will look for a solution for this also.

Do you mean "Expand/Collapse All" buttons for left side?

You can also use `Ctrl+P` fuzzy finder.

ell1e, 2020-10-08:
I just added a quick mockup image, I hope that will help!

 `CTRL+P` sounds useful, but sadly doesn't seem to pick up on subfolder paths inside all of the source folders. E.g. As you can see, the file `/home/user/Develop/core.horse64.org/horse64/main.c` is opened up. However, if I type `compiler/scoperesolver.c` to open up `/home/user/Develop/core.horse64.org/horse64/compiler/scoperesolver.c`, that entry won't be listed. I suspect it just doesn't look in that path where `main.c` is at, if it did that, that'd be very useful!

rohanrhu, 2020-10-08:
> Edit: here's a quick mockup idea:
> 
> [image]

Ok. Nice idea. A button for "list all" behaviour and I can make a "filter" setting for that. So you can set it to list only certain file types.

ell1e, 2020-10-08:
Your frontend is really good-looking by the way. Other than these small difficulties browsing the files it looks way more promising than all other frontends I have tried, which usually tend to feel really outdated and horrible to use for some reason. So I really appreciate this project.

rohanrhu, 2020-10-08:
Thank you. You can also access to GDB shell with `tmux a -t gdb-frontend` command in terminal. (and you can use terminalless layout without `/terminal/` path in URL.)

You can set that ID (gdb-frontend) with `--terminal-id=ID` option.

rohanrhu, 2020-11-28:
Hi,

I added FileBrowser opener buttons for each directory in SourceTree.

[image]

I still have more feature ideas for SourceTree and planning them for `v1.0.0`.

closed 2020-11-28 by rohanrhu