rohanrhu/gdb-frontend #14: Suggestion: clicking the "run" icon (the bug that will start the debugging) should show up a dialog to prompt me for command line arguments with "Run", "Cancel" buttons instead of running directly

ell1e, 2020-10-08:
I discovered that clicking the "Run" icon (the bug that will start the debugging) just starts the program directly. However, I am debugging a compiler, so I need to submit arguments to it so it'll actually compile the test file instead of just quit instantly. I think the most natural way to do that would be if clicking that "Run" icon just prompted me with a popup dialog with run options like command line arguments, environment vars, etc with "Run", "Cancel" buttons to confirm and actually run and cancel out, respectively.

Alternatively, adding a "Launch Options" entry to the top-right hamburger menu icon would also be a good way where people could find & edit this, if you really don't want the "Run" icon to require one additional click to confirm to actually fully run it. (Which I suppose would be an understandable objection)

rohanrhu, 2020-10-08:
Hi, thank you for feedback. I will add a menu for setting run arguments. For now, you can use `run [arguments..]` command on GDB shell. (Once you use args in run commands, it remembers args in future runs with run button.)

ell1e, 2020-10-08:
Ah, cool! Thanks, that sounds very useful @ remembering the args on future runs

rohanrhu, 2020-10-12:
I added run arguments setting for run button.

[image]

closed 2020-10-12 by rohanrhu