rohanrhu/gdb-frontend #15: Security suggestion: prompt for password when running `./gdb-frontend`, and ask for whatever is set when browser connects

ell1e, 2020-10-09:
Sadly, at least firefox (and at least up until recently chrome, I'm not sure they fixed it yet) will allow any **external** website to connect  to any **local** http port, including something like gdb-frontend. There are plans underway to require a special HTTP header to be sent such that browsers will block it if it's not marked as a safe target, but again I don't think these are widely available just yet. So this means in theory external websites could take over gdb-frontend, and browse the filesystem, mess around with gdb, ... unless there is any sort of trivial measure, like a simple password set by the user (doesn't even need to be HTTPS or anything) to prevent this. This can also be a problem in a hot seat situation where multiple users work on one machine and not all ports are locked down by default, or a sandboxed program doesn't have access to most system resources but probably does to networking.

In overall, I think it'd be a lot safer with just a simple prompt for any short password, first set at launch on command line via `./gdb-forntend` and then it'd be required via basic http auth when a browser connects. The passworwd doesn't even need to be ever stored on disk or whatever, or be absolutely required - the launch script could allow an empty password as a skip for those who don't care.

rohanrhu, 2020-10-09:
Default bind address is `127.0.0.1`. (You can specify it with `--listen=ADDRESS` option: https://github.com/rohanrhu/gdb-frontend#--listenip--l-ip)

It is not possible to access someone from another IP address to your debugger. But we still need a password-protected interface.  So we can share our session as readonly with that feature.

Im adding this to [v1.0 Roadmap](https://github.com/rohanrhu/gdb-frontend/projects/2).

ell1e, 2020-10-09:
If you browse google.com in firefox, it can definitely access the debugger. (Like, google literally can. Or any other website) This is not commonly known, but most browsers allow this right now. Browsers allow accessing most local ports through *external* websites, so this is maybe more of a problem than you think.

rohanrhu, 2020-10-09:
Do you mean a browser tab can access to another one?

ell1e, 2020-10-09:
gdb-frontend doesn't even need to be open in a tab. Just any website you open can send HTTP requests to all your local network through your web browser via JavaScript.

The firefox bug for this is here: https://bugzilla.mozilla.org/show_bug.cgi?id=354493 (firefox doesn't seem to have any sort of permission or prevention for this in place at all right now)

The chromium bug for this is here: https://www.chromestatus.com/feature/5733828735795200 (as you can see they're testing that the connection is aborted if the localhost or network host http server doesn't respond with a certain opt-in HTTP header, but you can still get spammed with partial connections once that is in - and right now they're only testing it so this isn't actually enforced yet)

rohanrhu, 2020-10-09:
That is a horrible bug.

ell1e, 2020-10-09:
Yeah I agree, even worse that it is "by design". It's a reckless idea if you ask me, exposing all local servers like that. Basically, nothing you run without a password is safe from a modern web browser. I hate it too, but not much that can be done to change it since all browsers behave like that right now :woman_shrugging: so the best options are things like a password to keep snoopy javascript apps away

rohanrhu, 2020-10-09:
As I understand, you don't mean other apps that are running on localhost can access the debugger naturally?

ell1e, 2020-10-09:
Hmm that depends on the app, that might also be possible. But I mean just any random website you visit, since the web browser will happily forward that random website's javascript HTTP ajax requests to your localhost port without asking you. It's considered a "feature" :woman_facepalming:

rohanrhu, 2020-10-09:
It is interesting. I dont think it can be possible, just tried and got CORS error: `Access to XMLHttpRequest at 'http://127.0.0.1:5551/' from origin '...' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.`

Other localhost pages still can access the debugger but that means they are already running on your computer and they have full access on the computer already.

Can you produce it as accessing localhost from different hostname/address?

ell1e, 2020-10-09:
Ah interesting. So basically it is possible to reach it via HTTP but the browser will abort if it doesn't have a same origin header that allows any source? Maybe it's not as bad as I thought then.

Re-reading, it seems the issue of chromium also seems to be about one intranet site messing with another: https://www.chromestatus.com/feature/5733828735795200 so you might be correct that a google tab opened in a web browser couldn't fully operate a gdb frontend (and I was possibly wrong).

In that case you'd be right and this is mostly just a local app issue for e.g. flatpak sandboxing, but not nearly as bad as I thought

rohanrhu, 2020-10-09:
Actually, it also must be not possible to different ports can access each other. I thought it maybe only possible with applications with multiplexed ports to same hostname with different URL paths.

ell1e, 2020-10-09:
Where I came into contact with this is with MEGASync, which runs at a local port and even in private mode, mega.co.nz can happily speak with the local instance. I suspect the local MEGASync backend must send the according Same Origin headers then to allow it, so I guess only then access from a different (possibly external) host + port is allowed? Sorry for jumping to conclusions :sweat_smile: 

In any case, I would prefer to keep local apps out via password too if possible :blush:

rohanrhu, 2020-10-09:
Yes we need a password option for better debugger sharing experience 😃

ell1e, 2020-10-09:
I've been poking around the `run.py` code some more, I wonder if it might sense to also add an option to not run it in tmux? (Since that is another "sharing" mechanism that might otherwise need to be passworded maybe, I don't use tmux so I don't know) Could be useful for those like me who don't use it anyway, I just use the web frontend and the terminal directly in the web tab and that's it

rohanrhu, 2020-10-09:
Tmux is solving so many issues. Terminal sharing to web would be a problem without tmux. If you wonder about tmux sessions, different linux users can't access to other's tmux sessions.

rohanrhu, 2020-10-09:
I added password option. You can use as `--credentials=user:pass` or `-c user:pass`.

But it is not ok yet because WebSocket debug server is still passwordless. It will be done when I added credentials to WS server too.

rohanrhu, 2020-10-11:
I implemented a new WebSocket server into HTTP server. (https://github.com/rohanrhu/gdb-frontend/commit/33f20c11801def3026e203019174bff76d7d5683) So we are not using a different WS library and port no longer. Now we are using two ports. (Gotty port and HTTP server port)

HTTP Auth security is now done.

closed 2020-10-11 by rohanrhu