rohanrhu/gdb-frontend #16: (Not sure if bug?) Opening a .c file will sometimes not actually open it as a source code file tab

ell1e, 2020-10-09:
In the project of mine I'm trying to debug, when I open a specific .c file via "Load File" then instead of showing up as an additional source code tab (which is what I wanted) instead the tree view disappears. It almost seems like gdb-frontend thought it was a binary, but it is not - it's really just a very boring text file. Am I supposed to open source files via some other button, or is that a bug? In any case, it'd be cool if "Load File" actually detected a code file vs a binary and magically did "the right thing" for both which at least for this case doesn't seem to have worked out.

Here is the issue in a short clip where I open the .c file:

[image]

rohanrhu, 2020-10-09:
"Load File" button is different from the load icon in right side of "Sources" section title.

* "Load File" button is for loading executables.
* Sources --- Load icon is for opening any file in editor.

ell1e, 2020-10-09:
Maybe you should rename it to "Load Executable" then? Might also be worth finding some way to filter the file list to only executables one day (e.g. by probing the header of the listed files for ELF and other common formats)

rohanrhu, 2020-10-09:
Good idea. Renamed it to "Load Executable". Thank you for feedback. I will change icon also.

ell1e, 2020-10-09:
You are so quick :heart_eyes: :+1:

closed 2020-10-09 by ell1e

rohanrhu, 2020-10-09:
By the way, when you add breakpoint a line in `main()` and step over the line for first time, you will see all sources on the left side and you can use Ctrl+P fuzzy finder also.

It seems also we have a scrollbar style problem for Firefox, I will look to that.