rohanrhu/gdb-frontend #19: OSX support

mark2185, 2020-11-04:
Any chance we might get a `mac + lldb` version?

I tried modifying it and solving the issues, but ultimately stopped because `lldb` doesn't support the `-ex` flag and I thought you might have a better idea on how to circumvent that.

rohanrhu, 2020-11-05:
Hi,

LLDB is a different debugger. Do you have the requirements: `Python3`, `GDB with Python3` and `Tmux`?

What errors are you getting? You can access GDB shell after running GDBFrontend, with this command: `tmux a -t gdb-frontend`. Do you see any error in there?

mark2185, 2020-11-05:
Well, I tried replacing the path to the `gdb` binary in the code with the path to `lldb`, and changing `python` to `python3`, and replacing the path to the prebuilt `gotty` with an appropriate one, installed by `brew`.

But for example line `284` fails since `lldb` doesn't support the `-ex` flag. Then I tried starting the program with `-p 0` so it skips that line ( by using a random port ), but on line `331` it requires the same logic, and that's necessary for the `gdbfrontend` to load and run, and `lldb` doesn't support it.

So I thought you might have a better idea how to solve that.

rohanrhu, 2020-11-05:
I thought your GDB has no Python3. I found this: https://stackoverflow.com/questions/48962558/installing-gdb-with-python-3-on-mac-with-homebrew

LLDB is a different debugger. I have not used that before and dont know about is it a GDB fork or GDB-compatible debugger. So you can check your GDB for is it including embbedded-Python3.

rohanrhu, 2020-11-26:
What about this issue?

mark2185, 2020-11-26:
We can close it. I'd rather not install `gdb` on mac, but I will try this debugger out on linux!

Thanks for your efforts!

closed 2020-11-26 by mark2185