rohanrhu/gdb-frontend #22: move src to new path, breakpoints not work

ziyouchutuwenwu, 2021-01-01:
i made a demo project with debug info. in the original path, gdbfrontend works great.
if a move the whole project dir to new dir, i click "Load Executable", it shows the original src path, and when i click the src file appeared in the left panel, it popuped with error msg "Path not found".
then i click the "add source" button which is in the right corner of the "Sources" panel. select my source file, set breakpoints.

but when i click the "debug" button, the demo bin just run, no breakpoint hit, info looks like this
```sh
![image](https://user-images.githubusercontent.com/7458713/103437711-c4acd100-4c65-11eb-8d9a-3a8fa0e4f970.png)
```

ziyouchutuwenwu, 2021-01-01:
if i set src path manually, kdbg can know it, and if possible, would gdbfrontend find debug symbol files from relative path as first choice?

rohanrhu, 2021-01-01:
Hi, Thank you for your feedback. Breakpoints are setting by full paths for now. I will check it out to make it to work with relative paths.

Have a nice day.

grooverdan, 2022-01-12:
Related,

[image]

rohanrhu, 2022-01-13:
> Related,
> 
> MariaDB container for remote debuggging:
> 
> ```
> $  podman run -d --rm -p 2345  --cap-add CAP_SYS_PTRACE --name mdb-debug -e MARIADB_ALLOW_EMPTY_ROOT_PASSWORD=1   quay.io/mariadb-foundation/mariadb-debug:10.2
> $ podman port mdb-debug 2345
> 0.0.0.0:41021
> $   podman exec --user mysql -ti  mdb-debug gdbserver --attach  :2345 1
> Attached; pid = 1
> Listening on port 2345
> Remote debugging from host 10.0.2.100
> ```
> 
> I used a `target remote :41021` on the gdb command line. I explicitly running gdbfrontend from a checked out source directory: (this uses a gdb-12 debugger that actually fetches the debug info).
> 
> Then set the breakpoint, continued. Triggered with `$ podman exec -ti mdb-debug mysql -e 'set global innodb_buffer_pool_size=256*10*1024*1024'`
> 
> resulting in:
> 
> [image]
> 
> However the relative info in the debuginfo can't automatically translate to the PWD as its been changed.

Helloo, when you click a line gutter, GDBFrontend sets that breakpoint with absolute path of source file. I will find a solution for this.

You can start GDBFrontend on your remote machine and connect to that from your local machine instead of using `gdbserver`.

Since `gdbserver` limits some functionalities, I think running GDBFrontend on remote machine and connecting that from local machine is more useful. Soooo is there a special reason to use `gdbserver`?