rohanrhu/gdb-frontend #23: Will not open the executable

MReed1959, 2021-01-05:
I have a program, compiled in the Windows 10 Linux shell. When I navigate to the executable I get this error: "Path not found."
The 'Sources' frame on the left side opens all the correct directories, the file browser shows the executable.
It would also be nice if I could specify the executable on the command line.

MReed1959, 2021-01-06:
https://user-images.githubusercontent.com/31863225/103818376-0a82e400-502e-11eb-9f84-01c090a5b6fe.mp4

rohanrhu, 2021-01-06:
>  2021-01-06-14-35-00.c.mp4

If you want to load an executable, you must use "Load Executable" button on top. The button placed on Sources tab is for opening additional source files.

Also, you might want to use GDB shell so you can open it as `http://127.0.0.1:5551/terminal/`.

MReed1959, 2021-01-06:
Ok, thanks!  I'll try that.

MReed1959, 2021-01-06:
When I go to "http://127.0.0.1:5551/terminal" I get a "This path not
found." page. It only works when I use the "http://127.0.0.1:5551" URL

BUT - I just found the bug!

Thanks SO MUCH!

rohanrhu, 2021-01-06:
Path should be "/terminal/" there is a last "/" at the end.

closed 2021-01-06 by rohanrhu

MReed1959, 2021-01-07:
Oh! Ok. Thanks again!