rohanrhu/gdb-frontend #24: Update archlinux aur PKGBUILD and fix argument passthrough

Adk9p, 2021-01-15:
A reminder that the aur PKGBUILD needs an update :)

Also currently the way the PKGBUILD file is set up it installs a script into /usr/bin that calls gdbfrontend without passing through arguments, leading to many confusing `--help`s that just started the server :p

This is my modified PKGBUILD that a used to install 0.5.0
```bash
# Maintainer:  Dimitris Kiziridis <ragouel at outlook dot com>

pkgname=gdb-frontend-bin
pkgver=0.5.0.beta
_pkgver=0.5.0-beta
pkgrel=1
pkgdesc="An easy, flexible and extensionable GUI debugger"
arch=('x86_64')
url='https://oguzhaneroglu.com/projects/gdb-frontend'
license=('GPL3')
provides=('gdb-frontend' 'gdbfrontend')
depends=('gdb' 'python' 'tmux')
source=("${pkgname}-${_pkgver}.tar.gz::https://github.com/rohanrhu/gdb-frontend/archive/v${_pkgver}.tar.gz")
sha256sums=('4efcfde256f0e40ceb1603e2222033e240c5438b81b6b13054a428825ce38d9c')

package() {
  install -d "${pkgdir}/opt/" "${pkgdir}/usr/bin/"
  cp -aR ${pkgname%-bin}-${_pkgver} "${pkgdir}/opt/gdbfrontend"

  cat <<- '  EOF' | sed 's/^\s*//' > gdbfrontend.sh
    #!/usr/bin/env sh
    cd /opt/gdbfrontend/
    ./gdbfrontend "$@"
  EOF
  install -Dm755 gdbfrontend.sh "${pkgdir}/usr/bin/gdbfrontend"
}
```

rohanrhu, 2021-01-15:
Hi, Thank you for packaging. 🙂 I'm gonna update the README.md.

Adk9p, 2021-01-16:
Hi thanks for the update, though the issue with the argument passthrough still persists.

The script I provided did more then change the version number, it also added `"$@"` after the call to gdbfrontend to pass the arguments. Without this no matter what you put after `gdbfrontend` in the terminal the script will ignore then and call gdbfrontend end with no arguments.

rohanrhu, 2021-01-16:
I think this way is already ok 🙂

Same way is used in runner bash script:
https://github.com/rohanrhu/gdb-frontend/blob/2a5ae896b3a6d4d1832f58615334e866679aa30c/gdbfrontend#L1-L3

closed 2021-02-02 by rohanrhu