rohanrhu/gdb-frontend #25: Method of opening UI is awkard

awsdert, 2021-03-11:
I'd rather gdbfrontend open it's own window with the webpage embedded in it instead of manually opening a web tab, even with the control & click of the terminal it is still awkward, for instance what if I want to run gdbfrontend from Geany's terminal? I won't be able to click the link there I expect, much easier if it just auto opens a window with the URL already loaded, also makes it easier to end the debug session if it just ends with the window, otherwise the UI seems like a good substitute for gede which is currently not downloading from AUR for me

rohanrhu, 2021-07-08:
Thank you for your suggestion. I'm planning a standalone window mode with `v1.0`. You can look [Radmap of v1.0](https://github.com/rohanrhu/gdb-frontend/projects/2). I will do all of my ideas and your suggestions when I come back from military.

rohanrhu, 2021-09-04:
I've added the feature. It opens GDBFrontend on the browser automatically on startup. Also it will open too from WSL. Thank you for your feedback.

closed 2021-09-04 by rohanrhu