rohanrhu/gdb-frontend #26: gdbinit support

zzdts, 2021-04-01:
How to config  GDB when gdb-frontend start ?
It seems  not read .gdbinit file when i have  start with -w option?

rohanrhu, 2021-09-03:
I've added `--gdb-args="ARGS", -G "ARGS"` option. You can specify your `.gdbinit` with it.

closed 2021-09-03 by rohanrhu

VelorumS, 2025-05-01:
Note that gdb-frontend sets this:

```
set confirm off
set non-stop off
set pagination off
```

if anyone struggling to understand why their gdbinit isn't working.

What kind of problems the non-stop will cause?

rohanrhu, 2025-05-01:
> Note that gdb-frontend sets this:
> 
> ```
> set confirm off
> set non-stop off
> set pagination off
> ```
> 
> if anyone struggling to understand why their gdbinit isn't working.

You can still set them after the debugger is booted up.

> What kind of problems the non-stop will cause?

As far as I remember, GDB's non-stop mode is necessary for GDBFrontend's thread to be able to run while your app is running on debugger.