rohanrhu/gdb-frontend #31: macOS support?

p-vernaeckt, 2021-09-13:
Hi!

I tried giving gdbfrontend a test drive on my macOS laptop this morning, and was only treated with an error message:
 ```
GDBFrontend v0.6.1-beta
Traceback (most recent call last):
  File "/Users/x99/.pyenv/versions/3.9.5/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Users/x99/.pyenv/versions/3.9.5/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/x99/.pyenv/versions/3.9.5/lib/python3.9/site-packages/gdbfrontend/__main__.py", line 22, in <module>
    spec.loader.exec_module(run)
  File "<frozen importlib._bootstrap_external>", line 855, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/Users/x99/.pyenv/versions/3.9.5/lib/python3.9/site-packages/gdbfrontend/run.py", line 348, in <module>
    gotty = subprocess.Popen(
  File "/Users/x99/.pyenv/versions/3.9.5/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Users/x99/.pyenv/versions/3.9.5/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: './bin/gotty'
```

macOS Version: Big Sur 11.5.2
pip 21.2.4
python 3.8.5

Any help is greatly appreciated 😅

rohanrhu, 2021-09-13:
Looks like we need a gotty executable for macOS 🙂 I'm currently working on a new feature, I'm going to stash it and look this issue. 👀

rohanrhu, 2021-09-13:
I've added MacOS executable for Gotty.

Can you try `macos-compatibility` branch?: https://github.com/rohanrhu/gdb-frontend/tree/macos-compatibility

rohanrhu, 2021-09-13:
I will do stuffs about MacOS compatibility on `macos-compatibility` branch if it needs.

p-vernaeckt, 2021-09-13:
That was quick, thank you!

I launched gdbfrontend from the cloned repo (I was using brew), and I get this new error:

```
./gdbfrontend: line 10: realpath: command not found
python3: can't open file '/run.py': [Errno 2] No such file or directory
```

rohanrhu, 2021-09-13:
Ah looks like `realpath` is not available on Mac shell.. I'm looking.

Were you running with `python3 run.py` ? So is it working on Mac now?

rohanrhu, 2021-09-13:
Hi, I've added a custom `realpath` for Mac shell. Can you try the latest revision? https://github.com/rohanrhu/gdb-frontend/commit/04c4f1a8e352ef59f23e7b3fb247fca5cd74abf4

p-vernaeckt, 2021-09-13:
New error:
```
GDBFrontend v0.6.2-git
Traceback (most recent call last):
  File "/Users/x99/Desktop/gdb-frontend/./run.py", line 350, in <module>
    gotty = subprocess.Popen(
  File "/Users/x99/miniconda3/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Users/x99/miniconda3/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: './bin/gotty_macos'
```

Ran `chmod +x bin/gotty_macos`, got this one afterwards:
```
GDBFrontend v0.6.2-git
Traceback (most recent call last):
  File "/Users/x99/Desktop/gdb-frontend/./run.py", line 350, in <module>
    gotty = subprocess.Popen(
  File "/Users/x99/miniconda3/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Users/x99/miniconda3/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 86] Bad CPU type in executable: './bin/gotty_macos'
```

Replaced `bin/gotty_macos` with the one I had installed from (here)[https://github.com/yudai/gotty] using `brew install yudai/gotty/gotty`, got this:
```
GDBFrontend v0.6.2-git
Listening on 127.0.0.1: http://127.0.0.1:5551/
Open this address in web browser: http://127.0.0.1:5551/terminal/
Sending SIGKILL to PGID: 1582.
Stopped GDBFrontend.
```

rohanrhu, 2021-09-13:
Are you sure for the port is available? 🙄

Also can you send outputs of these commands:
```bash
ps -e | grep gdb
```
```bash
tmux ls
```
```bash
tmux a -t gdb-frontend
```

If you don't see anything, add `while True: pass` to end of the file: https://github.com/rohanrhu/gdb-frontend/blob/7b9d2c2cab01489d2f86c93b6e9021fe87bd897e/gdbfrontend.py#L69-L71
Like this:
[image]

and later, try again `./gdbfrontend` by ensuring the port is available.

If we can't catch it with this way, I will do something for you to get some errors from tmux session.

rohanrhu, 2021-09-15:
Hi, any progress? I'm not able to test it on macOS since i don't have a mac.

p-vernaeckt, 2021-09-15:
Hi! No progress yet, no time to test right now. But I will soon 😉

rohanrhu, 2021-09-15:
> Hi! No progress yet, no time to test right now. But I will soon 😉

Ok. Have a nice day 🙂

rohanrhu, 2022-10-01:
This issue must be fixed. Also there was this issue: https://github.com/rohanrhu/gdb-frontend/issues/37 it is fixed too.

closed 2022-10-01 by rohanrhu