rohanrhu/gdb-frontend #33: Add registers display support with x86 ISA？

0400H, 2021-11-07:
Just like gdbgui, it can display the common registers and vector registers(xmm ymm zmm, etc).

It is very unfortunately for gdbgui that has not been updated for a long time, making the registers unavailable.

Are there any plans to add this feature？

[image]

rohanrhu, 2021-11-07:
I will add a registers section in the next version. Thank you for your feedback. 🙂

For now you can add registers to Watches like `$rax`.

rohanrhu, 2021-11-09:
It is done. I will release it soon.

[image]

rohanrhu, 2021-11-09:
> Just like gdbgui, it can display the common registers and vector registers(xmm ymm zmm, etc).
> 
> It is very unfortunately for gdbgui that has not been updated for a long time, making the registers unavailable.
> 
> Are there any plans to add this feature？
> 
> [image]

You can try it from latest GIT revision. I will release `v0.9.0-beta` with some more features.

0400H, 2021-11-10:
> You can try it from latest GIT revision. I will release `v0.9.0-beta` with some more features.

ENV
```
  Ubuntu 18.04 Docker
  gcc/g++ 7.5.0
  gdb 8.1.1
  python 3.6.9
```

gdb-frontend can not work on "[Added Registers view](https://github.com/rohanrhu/gdb-frontend/commit/1518921b3f0b64d7d41b52c1dcf08395e9600a02)". Nothing and no response in the Web-UI.
gdb-frontend can work on "[Update information about zombie processes](https://github.com/rohanrhu/gdb-frontend/commit/f6d5987c6465479d8d5c22ceb6daa47c33d4f1a8)".

It is recommended to provide a base dockerfile to testing and contributing code.

rohanrhu, 2021-11-10:
> gdb-frontend can not work on "[Added Registers view](https://github.com/rohanrhu/gdb-frontend/commit/1518921b3f0b64d7d41b52c1dcf08395e9600a02)". Nothing and no response in the Web-UI. gdb-frontend [...]

Both revisions are same.

Sometimes, xterm.js may lock interactions on the window. In this situation, just focus the terminal that you ran `gdbfrontend` command and press enter for once, then it will get well.

> It is recommended to provide a base dockerfile to testing and contributing code.

One of our main requirements is platform-compatibility, so we just need to test revisions or different OSes/platforms. Recently I have started to test it on different distros on virtual machine. So Docker is not something we need.

If you want to contribute to GDBFrontend, you can join to Discord server. https://discord.gg/RyVY9MtB4S You can ask about development of the debugger on Discord.

Happy debugging! 🎉🎉🎉

rohanrhu, 2021-11-19:
This feature is done and released.

closed 2021-11-19 by rohanrhu