rohanrhu/gdb-frontend #35: Could separate the front-end and module parser？

0400H, 2021-11-12:
Is it possible to separate the front-end and module parser so that they can run independently？
After separating it, the module parser can be used for other projects.

Just like [tmux+gdb](https://github.com/ficoos/tgdbdb)

rohanrhu, 2021-11-12:
It is actually usable with importing `gdbfrontend` module in any GDB session when you install it with PIP but I must do something for that usage.

Can you explain more?

0400H, 2021-11-12:
、

> It is actually usable with importing `gdbfrontend` module in any GDB session when you install it with PIP but I must do something for that usage.
> 
> Can you explain more?

For example
```
https://github.com/rohanrhu/gdb-frontend/tree/master/api
https://github.com/rohanrhu/gdb-frontend/blob/master/plugins/hello/hello.py
```

```
import importlib
import json

import plugin
import api

gdb = importlib.import_module("gdb")

class HelloPlugin(plugin.GDBFrontendPlugin):
    def __init__(self):
        plugin.GDBFrontendPlugin.__init__(self)

    def loaded(self):
        gdb.events.new_objfile.connect(self.gdb_on_new_objfile)

    def unloaded(self):
        gdb.events.new_objfile.disconnect(self.gdb_on_new_objfile)

    def gdb_on_new_objfile(self, event):
        print("[HELLO] GDB Event: new_objfile:", event)
    
    def event(self, client, event, message):
        print("[HELLO] GDBFrontend Event Client#%d: %s: %s" % (client.client_id, event, message))

        for _client in api.globalvars.httpServer.ws_clients:
            _client.wsSend(json.dumps({
                "event": "hello",
                "replt_to": event
            }))
```

Maybe can remove `http` parts of `hello.py`(module parser), just keep I/O Interface.
Establish unit testing for module parser.
Rewrite the `API/HTTP` in python to load all module parser.

Just like
```
|- api
  |- module
       |- __init__.py
       |- modules
       |- test
  |- http
|- ...
```

closed 2021-11-19 by rohanrhu