rohanrhu/gdb-frontend #38: Can't scroll up the terminal

crucio55, 2021-11-24:
I'm trying out `gdbfontend` for the first time, debugging an executable whose source files do not sit on the machine that runs `gdb` (and thus the executable).
I can't scroll up the terminal, yet it's the only place where I can see the program's output - and also the only place where I can add breakpoints.

There aren't any `tmux` settings for the user running `gdbserver`, by the way.
This is what it looks like:
[image]

rohanrhu, 2021-11-24:
With default settings, tmux's `mouse` setting is enabled. This means the scrollbar of the terminal doesn't work. Instead, we can scroll with mouse wheel.

I have some questions to understand the issue:

1. Are you trying to scroll with mouse wheel?
2. Are you able to type something on terminal even though you can't do scroll?

crucio55, 2021-11-25:
1. Yes, I used the mouse wheel
2. Yeah, I can type in it

rohanrhu, 2021-11-25:
> 1. Yes, I used the mouse wheel
> 2. Yeah, I can type in it

Interesting... What browser are you using?

crucio55, 2021-11-25:
> > 1. Yes, I used the mouse wheel
> > 2. Yeah, I can type in it
> 
> Interesting... What browser are you using?

Vivaldi

rohanrhu, 2021-11-25:
> > Interesting... What browser are you using?
> 
> Vivaldi

It is really interesting. Can you try it on another browser? When you do scroll, is terminal cursor being moved?

rohanrhu, 2021-11-27:
@crucio55 Hello?

crucio55, 2021-11-28:
> > > Interesting... What browser are you using?
> > 
> > 
> > Vivaldi
> 
> It is really interesting. Can you try it on another browser? When you do scroll, is terminal cursor being moved?

Tired out Firefox, seems the same. With ctrl+b (tmux's default control key bind) i manage to scroll the terminal with the arrows and page up/down - but I can't exit that mode (exiting it is done by Esc - but in gdbfrontend it just removes focus from the terminal)

rohanrhu, 2021-11-28:
> Tired out Firefox, seems the same. With ctrl+b (tmux's default control key bind) i manage to scroll the terminal with the arrows and page up/down - but I can't exit that mode (exiting it is done by Esc - but in gdbfrontend it just removes focus from the terminal)

Ok. I will check it out.

I have some questions:

* What is your `tmux` version?
* Can you do `tmux set-option mouse on` and try again on the terminal?

crucio55, 2021-12-07:
Hi, sorry for the delay!

> * What is your `tmux` version?
> * Can you do `tmux set-option mouse on` and try again on the terminal?

* `tmux` version is 2.1, though I also have 2.3a on the machine that isn't the default version for this user
* I tried with `set-option mouse on` in `.tmux.conf`, the results are the same in 2.1 - but scrolling **does** now work when using 2.3a!!!

rohanrhu, 2021-12-07:
> * I tried with `set-option mouse on` in `.tmux.conf`, the results are the same in 2.1 - but scrolling **does** now work when using 2.3a!!!

Nice! It sets tmux's `mouse` option to `on` as default on startup. Must be something wrong with `tmux 2.1`.

Let's add a notice about minimum tmux version or buggy tmux versions to README.md.

* What is your distro? Where did you install newer tmux?
* Is there any problem now?

crucio55, 2021-12-07:
> * What is your distro? Where did you install newer tmux?

Fedora 23 (ancient indeed) - with the new `tmux` installed manually from source

> * Is there any problem now?

Yeah, I keep getting these in the terminal:
[image]

rohanrhu, 2021-12-07:
> Yeah, I keep getting these in the terminal: [image]

Are you using `--verbose` option? Actually without verbose option, you must see only `gdb.error: value has been optimized out` part of this. It means your code is compiled with an optimization flag (like `-O3`).

For debugging, you should compile your code with zero optimization and debug symbols flags: `-O0` and `-g`.

Like this:
```bash
gcc -o app app.c -O0 -g
```

If you see this traceback even if you don't use `--verbose`, I will avoid this traceback reporting.

crucio55, 2021-12-07:
> Are you using `--verbose` option? Actually without verbose option, you must see only `gdb.error: value has been optimized out` part of this. It means your code is compiled with an optimization flag (like `-O3`).
> 
> For debugging, you should compile your code with zero optimization and debug symbols flags: `-O0` and `-g`.
> 
> Like this:
> 
> ```shell
> gcc -o app app.c -O0 -g
> ```
> 
> If you see this traceback even if you don't use `--verbose`, I will avoid this traceback reporting.

No, I'm not using the `--verbose` option.

Optimizing with `-O3` was indeed done, and some values being optimized out is a given - but getting the tracebacks without explicitly specifying they're wanted with `--verbose` is less nice.

rohanrhu, 2021-12-07:
> Optimizing with `-O3` was indeed done, and some values being optimized out is a given - but getting the tracebacks without explicitly specifying they're wanted with `--verbose` is less nice.

Yes, I will reduce unnecessary warnings.

rohanrhu, 2021-12-13:
I reduced unnecessary warnings. It will be come with v0.10, also you can get the latest revision from GIT until the release.

For tmux problem I will add a "minimum tmux version" note to README.md.

Thank you for reporting!

closed 2021-12-13 by rohanrhu