rohanrhu/gdb-frontend #40: Maybe something other than tmux is available for Windows?

SeanTolstoyevski, 2021-12-06:
GDB can be used on Windows with MinGW projects. There is no problem.
But **Tmux** is not used, compiled for Windows, or nobody has tried it.
Is there another alternative to tmux for Windows?
Maybe we can use it on Windows. This way GDB Frontend will also support Windows.

rohanrhu, 2021-12-06:
The problem is not terminal multiplexing about Windows. I can make a terminal multiplexer for Windows but it is not the main problem.

Main problem is GDB's Windows version's main-thread is being blocked during application runtime. This situation maybe changed/solved in new versions of GDB, I've not checked it yet but I will check it out soon.

rohanrhu, 2021-12-06:
By the way, I had talked with GDB devs, they just said me it is possible but hard to make and nobody wants to implement it yet.

SeanTolstoyevski, 2021-12-07:
Thank you for your answer, @rohanrhu.

You can close this issue off if you wish.

closed 2022-10-20 by rohanrhu