rohanrhu/gdb-frontend #41: Failure with Python 3.6: unexpected keyword argument 'capture_output'

chipbuster, 2021-12-14:
Hey there! I'm running gdbfrontend v0.10.0-beta on a system with Python 3.6. When attempting to run the `gdbfrontend` command, it crashes with the following output:

```
GDBFrontend v0.10.0-beta
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/u/theksong/.local/lib/python3.6/site-packages/gdbfrontend/__main__.py", line 22, in <module>
    spec.loader.exec_module(run)
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/u/theksong/.local/lib/python3.6/site-packages/gdbfrontend/run.py", line 308, in <module>
    proc = subprocess.run([gdb_executable, "--configuration"], capture_output=True)
  File "/usr/lib/python3.6/subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
TypeError: __init__() got an unexpected keyword argument 'capture_output'
Sending SIGKILL to PGID: 28463.
Stopped GDBFrontend.
```

I believe the issue is at [run.py:308](https://github.com/rohanrhu/gdb-frontend/blob/918d63311320e038654fd9dc2d21e0e2738d7223/run.py#L308). The `capture_output` argument [is not supported by subprocess in Python 3.6.](https://docs.python.org/3.6/library/subprocess.html#subprocess.run). Replacing it with the following line (which I lifted from stackoverflow) gets rid of the error for me:

```
proc = subprocess.run([gdb_executable, "--configuration"], stdout=subprocess.PIPE, stderr=subprocess.PIPE)
```

I can open a PR for this if you'd like, but since it's only one line I figured it might be easier for you to deal with it.

rohanrhu, 2021-12-14:
Hi, fixed. Thank you for reporting 🙂 Is there another issue with Python 3.6? I'm gonna release v0.10.1-beta if it is ok.

rohanrhu, 2021-12-14:
I released a new version with this fix. Thank you again 🙂

rohanrhu, 2021-12-14:
If you see another issue and know how to fix, please feel free to send a PR.