rohanrhu/gdb-frontend #42: gdb prompt is behind bottom status bar

grooverdan, 2022-01-13:
[image]

rohanrhu, 2022-01-13:
Oh this problem 😔 I have yet to see this problem on my computer 🤭 but someone however has this problem. I will fix it soon.

I have some questions:
* What is your browser?
* What is your screen resolution and screen scale?

Thank youuuu soo much for feedback. 😊

grooverdan, 2022-01-13:
firefox 95.0.2
1920 x 1080 @ 100%

rohanrhu, 2022-01-28:
> firefox 95.0.2
> 1920 x 1080 @ 100%

I just checked it out with someone has this problem. It is happening on Firefox's Linux version. It is related to Xterm.js (looks like it has a bug) and Firefox on Linux.  Sooo I will look for a solution, you can use Chrome/Chromium instead until resolving this issue.

With Chrome you can run the debugger as a standalone app, so you will have more vertical space to work :)

```bash
chromium --app=http://127.0.0.1:5550/terminal/
```

I know some people don't want to use Chrome for some privacy reasons but since Chromium is also open source and good for privacy, you can use it if you consider this reason to use Firefox.

This opens the debugger as a standalone app. It looks like this:

[image]
[image]

grooverdan, 2022-01-31:
Thanks for looking into detail as to the cause of this and the alternative

rohanrhu, 2022-01-31:
> Thanks for looking into detail as to the cause of this and the alternative

Thank you tooo.

rohanrhu, 2022-12-18:
I'm closing this issue with the latest version xterm.js is up to date. If anyone has this issue again, please re-open.

closed 2022-12-18 by rohanrhu