rohanrhu/gdb-frontend #50: GDB executable not found due to $PWD is modified by gdbfrontend

RafaelLeeImg, 2022-04-13:
The command I ran is:
```bash
$ gdbfrontend -D -G "iceprog/iceprog -n -x ~/.gdbinit"
```
The problem is the executable iceprog/iceprog cannot be found
The working directory is "~/.local/lib/python3.7/site-packages/gdbfrontend" after I ran the command, thus, the executable "iceprog/iceprog" cannot be found.
Temporary solution is:
```bash
gdbfrontend  -D -G "`pwd`/iceprog/iceprog -n -x ~/.gdbinit"
```

There may be 2 solutions, first, pass $PATH environment variable to gdb; second, retain the working directory same as where the user runs the gdbfrontend.

rohanrhu, 2022-04-13:
Hiii @RafaelLeeImg thank youu for for reporting. I'm looking.

rohanrhu, 2022-04-13:
Hiii again @RafaelLeeImg 

Can you provide these informations?:
* GDBFrontend version
* OS

rohanrhu, 2022-04-13:
> The working directory is "~/.local/lib/python3.7/site-packages/gdbfrontend" after I ran the command

This sounds very interesting, because with `v0.11.0-beta` it must be setting the working directory to current directory that you run it.

Are you sure for you are using `v0.11-beta`?

RafaelLeeImg, 2022-04-14:
Yesterday, I installed gdbfrontend with pip, the version is:
gdbfrontend                        0.11.0
Operating System: Debian GNU/Linux bookworm/sid x86_64
Kernel version: 5.16.0-3-amd64

Today I also tried commit 9fba9fc - (2 weeks ago) Bump version v0.11.0-beta, the problem is partly solved but not fully.
[9fba9fce](https://github.com/rohanrhu/gdb-frontend/commit/9fba9fce808934a042dd6940225d99d329f2afe9)
I checked the code in run.py, it will change dir to the working directory, but that happens after the gdb args are parsed. Leaving the executable not found. Change directory should be done before invoke gdb.
```python
" \"cd " + config.WORKDIR + "\"" +
```

rohanrhu, 2022-04-14:
@RafaelLeeImg I removed changing directory to GDBFrontend directory in runner.

Can you try try this revision the latest revision? https://github.com/rohanrhu/gdb-frontend/commit/022dd9ad29b1d2820ae8de6a4b2c95c8029d5440

rohanrhu, 2022-04-20:
Fixed with [Remove module __main__ chdir for /usr/bin script](https://github.com/rohanrhu/gdb-frontend/commit/dcb804bbf579bf4a3140b96f7ea337b0404d0042)

There was a missing issue with `/usr/bin` run script, I just found out it and fixed.

Also `v0.11.2-beta` is released with this fix.

closed 2022-04-20 by rohanrhu