rohanrhu/gdb-frontend #54: cannot read file in demo?

likyh, 2022-07-24:
```cpp
#include <stdio.h>
int main(){
    int a, b;
    scanf("%d%d", &a, &b);
    printf("%d\n", a+b);
    return 0;
}
```

got:
[image]

rohanrhu, 2022-07-24:
When you use `scanf()` and similars, OS sends `SIGTTIN` to your process and GDB shell and your app can't use the same TTY like this. Your app must use a different TTY than the one that your GDB shell is on.

# Simplest way to debug apps that take inputs on TTY

## Split terminal and run your app in new terminal
[image]

## Attach GDB to your process
[image]

## When GDB attached to your process, your process will be interrupted immediately, **just click to continue**
[image]

## Add a breakpoint after the call that gets input from TTY and enter an input and press enter
[image]

That's all.

closed 2022-07-24 by rohanrhu

GitMensch, 2023-05-22:
That's very useful - What do you think about adding this to the tutorials and link that in README?

rohanrhu, 2023-06-13:
> That's very useful - What do you think about adding this to the tutorials and link that in README?

Thank you @GitMensch I will add it to docs and link in README.md.