rohanrhu/gdb-frontend #55: gdb main thread is bloocking ...

baiwfg2, 2023-02-19:
Thanks for making gdb so convenient. I've met with two problems.

One is when I'm debugging mariadbd program whose binary size is 195M, but show "gdb main thread is bloocking". I don't know what it means

another problem is that when I step over or step into, it's really slower than using gdb on the commandline. Can it be faster ?

[image]

rohanrhu, 2023-02-19:
Hi, thank you for your feedback.

Can you provide this informations?
* What are your OS and GF version?
* How often do you see this warning?
* How long does it take for each step?

baiwfg2, 2023-02-20:
@rohanrhu  Hi. My OS is :
Linux fedora 5.15.16-100.fc34.x86_64 #1 SMP Thu Jan 20 16:34:27 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

I've found that the moment I load executable such as mariadbd, this warning will often pop out

But if I load light-weight program such as postgres, my custom program, I don't see this warning and Step over/step into is fast enough.

So the problem seems to be why loading a big debug-version program has such issue ?

rohanrhu, 2023-02-20:
Thank youu @baiwfg2

All of GF functions are working on GDB's main thread so something must be taking long time for big applications like MariaDB and MySQL contexts.

There are two things:
* Backend functions
* Frontend rendering functions for each step

For example ProcessManager rendering is well-optimized but SourceTree is not optimized as well as ProcessManager.

Things I will improve:
* Backend functions
* Better rendering for SourceTree and inspector components

But do you feel any freezing in UI for each step? I debug some big applications like Godot Engine and I never felt like frontend is freezing or an annoying blocking main thread warning for softwares I debugged but SourceTree and other components must render like ProcessManager's way. So it is something I will improve as next thing to do.

Also I must test MariaDB and improve things that take long time in backend. MySQL is known as a bit memory hungry, it must have so many variables in each context and resolving them in backend (to show in frontend) must be taking some long time. **I will improve them to a way that is more iteration-ish.**

**The most important question for now is:**
Are you feeling like frontend/browser is freezing for each step. If not, I will focus on backend at first.

baiwfg2, 2023-02-21:
not freezing for each step.  Roughly frequent

rohanrhu, 2023-02-21:
> not freezing for each step. Roughly frequent

Oh good.. I was thinking about SourceTree but then it is good as I thought. At least it has some optimizations rather than rendering optimizations.

Then I will look at backend and optimize things for huge codebases.

Btw could you give your hardware details?

baiwfg2, 2023-02-22:
Actually I start a virtual fedora machine with virtualbox on my windows 10. The virtual machine is equipped with 2 CPUs, 5G RAM. 

Before I start GF,
```
[root@fedora data]# ps axu | grep maria
root        4298  1.0  2.6 1434460 132228 pts/1  Sl   21:48   0:01 ./sql/mariadbd --datadir /data2/mariadb-data/ --user root --port 3307
root        4341  0.0  0.0   6140   852 pts/2    S+   21:50   0:00 grep --color=auto maria
[root@fedora data]# free -m
               total        used        free      shared  buff/cache   available
Mem:            4916         858        2120           1        1936        3770
Swap:           4915           0        4915
```

After I start GF,
`gdbfrontend -l 192.168.0.104 `
```
[root@fedora ~]# ps aux | grep gdb
root        4352  0.0  0.0   6904  3156 pts/4    S+   21:50   0:00 /bin/bash /usr/local/bin/gdbfrontend -l 192.168.0.104
root        4353  0.3  0.3  21864 17856 pts/4    S+   21:50   0:00 python3 -m gdbfrontend -l 192.168.0.104
root        4354  0.0  0.2 169188 13660 pts/4    Sl+  21:50   0:00 python3 -m gdbfrontend -l 192.168.0.104
root        4366  0.0  0.0   6904  3140 pts/12   Ss+  21:50   0:00 bash -c gdb  -ex "python import sys, os; sys.path.insert(0, \"/usr/local/lib/python3.9/site-packages/gdbfrontend\"); import config, json, base64; config.init(); config.setJSON(base64.b64decode(\"eyJURVJNSU5BTF9JRCI6ICJnZGItZnJvbnRlbmQiLCAiV09SS0RJUiI6ICIvcm9vdCIsICJCSU5EX0FERFJFU1MiOiAiMTkyLjE2OC4wLjEwNCJ9\").decode()); import gdbfrontend"; read
root        4368 42.4  7.7 1033164 389804 pts/12 Rl+  21:50   0:26 gdb -ex python import sys, os; sys.path.insert(0, "/usr/local/lib/python3.9/site-packages/gdbfrontend"); import config, json, base64; config.init(); config.setJSON(base64.b64decode("eyJURVJNSU5BTF9JRCI6ICJnZGItZnJvbnRlbmQiLCAiV09SS0RJUiI6ICIvcm9vdCIsICJCSU5EX0FERFJFU1MiOiAiMTkyLjE2OC4wLjEwNCJ9").decode()); import gdbfrontend
root        4373  1.0  0.4 260864 24844 pts/12   Sl+  21:50   0:00 gdb -ex python import sys, os; sys.path.insert(0, "/usr/local/lib/python3.9/site-packages/gdbfrontend"); import config, json, base64; config.init(); config.setJSON(base64.b64decode("eyJURVJNSU5BTF9JRCI6ICJnZGItZnJvbnRlbmQiLCAiV09SS0RJUiI6ICIvcm9vdCIsICJCSU5EX0FERFJFU1MiOiAiMTkyLjE2OC4wLjEwNCJ9").decode()); import gdbfrontend
root        4460  0.0  0.5 632064 26428 pts/13   Ss+  21:50   0:00 gdb -ex python import sys, os; sys.path.insert(0, "/usr/local/lib/python3.9/site-packages/gdbfrontend"); import config, json, base64; config.init(); config.setJSON(base64.b64decode("eyJURVJNSU5BTF9JRCI6ICJnZGItZnJvbnRlbmQiLCAiV09SS0RJUiI6ICIvcm9vdCIsICJCSU5EX0FERFJFU1MiOiAiMTkyLjE2OC4wLjEwNCJ9").decode()); import gdbfrontend
root        4462  0.0  0.0   4440  3244 pts/13   S+   21:50   0:00 tmux a -t gdb-frontend -c /root
root        4617  0.0  0.0   6140   856 pts/8    S+   21:51   0:00 grep --color=auto gdb
[root@fedora ~]# free -m
               total        used        free      shared  buff/cache   available
Mem:            4916        1148        1830           1        1936        3480
Swap:           4915           0        4915
```
Then I open browser and attach mariadbd process, quickly the blocking information is popped out...

rohanrhu, 2023-02-22:
Thank you so much @baiwfg2 😊 I will fix it soon in next release.