rohanrhu/gdb-frontend #56: FeatReq: Close windows in right sidebar

RenHoekNL, 2023-02-21:
The right sidebar has a number of windows, however my screen is a little small on my laptop.

When I step through my program I want to keep an eye on the variables, but I can't see those without scrolling the sidebar down a bit.

Then I step, and the sidebar window is reset again to the top. Again, I can't see my variables anymore.

Fixes:
- Allow me to close windows I don't need, e.g. Disassembly
- Keep the windows position where it is when stepping

Ideally both these changes are put in. Thanks in advance.

rohanrhu, 2023-02-21:
Hi @RenHoekNL thank you for your feedback.

GF got a feature about that it must keep your current scroll position on each step.

It must act like this:
https://user-images.githubusercontent.com/1125150/220482343-46cde0c0-198b-4817-bb7d-851db5b2d310.mp4

Seems like it is not working for you 🙁

What is your browser? If it is not Chrome can you try it on Chrome?

RenHoekNL, 2023-02-22:
Can confirm it 'works' in Chrome. I'm using Firefox 110.0 (64-bit).

It does jump the sidebar around in Chrome as well, at least it stick to the part I was using. It doesn't just stay where I was though and when I'm deep in structs I still have to scroll back.

It could use some TLC.

rohanrhu, 2023-02-22:
> Can confirm it 'works' in Chrome. I'm using Firefox 110.0 (64-bit).

Oh good... Just use Chrome till I fix it for the other browser (btw what is that? Firefox?)

> It does jump the sidebar around in Chrome as well, at least it stick to the part I was using. It doesn't just stay where I was though and when I'm deep in structs I still have to scroll back.

It expands all previously opened structs right?

sgbeal, 2024-02-21:
Please consider this another request for the ability to remove unneeded right-side widgets. i literally never need the Registers window and it takes up a LOT of space which i'd much rather have for the Variables. i can use the browser's Inspector to remove the Registers widget, but then the Variables has a fixed max size so won't grow to fill that space. i can get around that by removing the max-height from `div.VariablesExplorer_content` it and adding `resize:vertical` to that element, but this is a tedious workaround.

rohanrhu, 2024-02-21:
> Please consider this another request for the ability to remove unneeded right-side widgets. i literally never need the Registers window and it takes up a LOT of space which i'd much rather have for [...]

Hi @sgbeal my brain is really burning because of working and some other things but I'll find a time for these things soon.

I'll make right bar sections optionally enabled/disabled, I'll also make right bar's variables explorer dynamic-sized.