rohanrhu/gdb-frontend #60: Problems encountered during trying out gdb-frontend

felixendres, 2023-12-11:
Hi rohanrhu,

this project looks quite promising. I tried it out today and ran into some troubles. Since you've mentioned you'd like testers, I guess you will not mind feedback.

- I ran some unit tests within gdbfront 0.11.3. They run fine under gdb, but got a sigsev in gdbfront.
- I set a breakpoint on line 32. gdbfront shows me it's in line 32 (red dot is in line 32 and breakpoint overview says it's in line 32). gdb (info breakpoints) says it's on line 33. When run it stops at line 33.
- Pasting a filename into the Ctrl+o dialog gives an error "not a directory". I found Ctrl+p now, but that was still confusing.
- Could the sections on the right (Disassembly, Watches, Breakpoints, ...) be collapsible? I often don't care for, e.g., Disassembly.

Anyway, despite the glitches it's impressive. Keep up the good work.
Best,
Felix

rohanrhu, 2023-12-11:
Hello, thank you for your feedback.

* What is having SIGSEGV exactly? Your app or GDB process?

* In DWARF debug info some source lines are not "stoppable at" or some expressions are like "ineffective":
For example,
```c
/* L:10 */ struct meow meow;
/* L:11 */ meow.count++;
/* L:12 */ meow.type = MEOW_TYPE_CUTE;
```
in this example if you add a breakpoint at line `10` your "actual breakpoint line" will be line `11`, because `struct meow meow;` line is ineffective (or nexecutable); it is telling the compiler to add `PUSH ...` instructions as much as `sizeof(struct meow)` in total size, so it is not a line that's being executed in runtime like others.

* Ctrl+O is not for loading executables. It is a "open source file" dialog. You can load executables with Load Executable button or like `file meow.exe` on GDB shell.
* Ctrl+O dialog doesn't support file path, you can enter a folder's path and select the file.
* Oh collapsing feature for right bar sections is a feature in my mind however I don't have time for that. That would be good if you can add that feature.

Thank you for your nice words.

felixendres, 2023-12-12:
Thank you for your answers. Unfortunately, I couldn't reproduce the sigsev anymore today, so I am not sure about what it was.

closed 2023-12-12 by felixendres