rohanrhu/gdb-frontend #61: macOS Thread no longer exists

NanShanJi, 2024-01-13:
Hello, I encountered an accident while using macOS
I'm sorry to bother you, but I really can't solve it
Installing GDB: https://formulae.brew.sh/formula/gdb#default

### system information：
macOS13.2.1 (22D68)
Intel Core i5

### terminal python：
[image]

### gdb-frontend python：
[image]
(I have been looking for a solution to why Python is 3.1, but I couldn't find it)

### start gdb-frontend：
[image]

### issue：
[image]

rohanrhu, 2024-01-13:
Hi, thank you for your feedback.

I know this issue, it appeared with one of recent versions; but it must not cause any problem in terms of functionality as I know.

Are you having any problem? Could you explain detailed the problem if you are having except seeing this exception message?

(Also I noticed that a known bug about the terminal height (it is being under the bottom bar) is still happening. I'll look at that too.)

NanShanJi, 2024-01-14:
> Hi, thank you for your feedback.
>
> I know this issue, it appeared with one of recent versions; but it must not cause any problem in terms of functionality as I know.
>
> Are you having any problem? [...]

Thank you very much for your reply
Yes, it does not affect usage, so I have decided not to solve it anymore
I would like to consult with you
How is the gdb package imported?
In gdb frontend, it is possible to avoid errors
[image]
If I use a. py file alone to write, I will report an error and cannot find the package
[image]

rohanrhu, 2024-01-14:
> Yes, it does not affect usage, so I have decided not to solve it anymore

Good, I think it is an unnecessary print of a well-handled (like an existence check) exception.

> I would like to consult with you
> How is the gdb package imported?
> In gdb frontend, it is possible to avoid errors

Thank you, I would like that since I don't have any time to work on the debugger. You can send pull requests.

The `gdb` module is available in GDB's Python runtime.

As I can remember, the reason that I do `importlib.import_module("gdb")` for `gdb` module, is for avoiding linter errors on VSCode. Because linter's Python doesn't have `gdb` module. But linters have des description line attributes too, we can use them too for this purpose but it would be linter-specific and combining multiple attributes for different linters might not work properly, idk.

NanShanJi, 2024-01-14:
> > 是的，不影响使用，所以决定不再解决了
> 
> 很好，我认为这是一个处理良好（如存在检查）异常的不必要的打印。（但我可能有点错，可能存在一个不影响使用的小问题。）
> 
> > 想咨询一下
> > gdb包是如何导入的？
> > 在gdb前端，可以避免错误
> 
> 谢谢，我希望如此，因为我没有时间在调试器上工作。您可以发送拉取请求。
> 
> 该`gdb`模块在 GDB 的 Python 运行时中可用。
> 
> 我记得，我`importlib.import_module("gdb")`为`gdb`模块做的原因是为了避免 VSCode 上的 linter 错误。因为linter的Python没有`gdb`模块。但是 linter 也有 des 描述行属性，我们也可以将它们用于此目的，但它是特定于 linter 的，并且为不同的 linter 组合多个属性可能无法正常工作，idk。

Thank you so much
I think all my questions have been resolved
This project has helped me a lot

rohanrhu, 2024-01-14:
> Thank you so much
> I think all my questions have been resolved
> This project has helped me a lot

Thank you too. You are using x86_64 macOS, is GDBFrontend working well on x86_64 macOS?

NanShanJi, 2024-01-14:
Yes, it can run normally 😄

rohanrhu, 2024-01-14:
> Yes, it can run normally 😄

So good. Thank you.

closed 2024-04-02 by rohanrhu