rohanrhu/gdb-frontend #65: Error message when running on WSL2 Linux

kingsumos, 2024-05-16:
Hi,

When starting gdbfrontend on WSL2 Linux the following error message is shown:
```
sh: /mnt/c/windows/system32/rundll32.exe: No such file or directory
```
Basically this is caused by run.py:363 script, which incorrectly detects a Windows system due to a "microsoft-standard-WSL2" string appended to the kernel version... For instance:
```
user@97b155e8a200:/usr/lib/python3.6/site-packages/gdbfrontend> python3
Python 3.6.15 (default, Sep 23 2021, 15:41:43) [GCC] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> platform.uname()
uname_result(system='Linux', node='97b155e8a200', release='5.15.146.1-microsoft-standard-WSL2', version='#1 SMP Thu Jan 11 04:09:03 UTC 2024', machine='x86_64', processor='x86_64')
>>> CTRL+D
user@97b155e8a200:~/git/template/prod> uname -r
5.15.146.1-microsoft-standard-WSL2
```
So today it is required to use --dontopenuionstartup when using WSL2

Thanks!
Sumo

rohanrhu, 2024-05-16:
Hi, thank you for your feedback. 🙂 I don't have time to fix it right now, it would be so good if you can fix it and make a PR.

MS must've been changed something on WSL 2. It was working fine previously but I don't use Windows for long time, I use GDBFrontend on Docker mostly.

rohanrhu, 2024-05-31:
Fixed.

closed 2024-05-31 by rohanrhu