rohanrhu/gdb-frontend #67: support for rr debugger

vimkim, 2024-07-18:
It will be amazing to support rr (record and replay) debugger.

Adding a 'reverse step' and 'reverse next' button would be amazing too!

rohanrhu, 2024-07-18:
You can do it in the same way as you do with only GDB, GDBFrontend delivers a terminal emulator at the bottom but buttons would be so nice. I think you mean packaging `rr` into the GDBFrontend distribution? How could we deliver the best solution for this? I don't know anything about rr.

vimkim, 2024-07-18:
Oh I see. I wish you to take a look at rr debugger one day (it works on WSL too).

RR is such an amazing debugger and I use it all the time. It is a game changer.

GDB's reverse next feature is quite limited and you can do it only for a small program. If you try to record a huge program such as mysql executable it burst out and breaks.

RR debugger, on the other hand, was made by Mozilla for debugging large executables like firefox browser.
It just feels like a miracle since **you can reverse the whole program state back from the end to the beginning**, even if you run firefox with it.

And the reason I'm suggesting you to implement it is that, in their documentation, it says

>rr implements the standard gdb server interface, and also can present the gdb command-line interface. This means it can be used in integrated development environments (IDEs) which support GUI debugging based on gdb/gdbserver.

https://github.com/rr-debugger/rr/wiki/Using-rr-in-an-IDE

I'm not into details but it seems  it might be easy to integrate rr into your debugger, with just a little tweak.

I see other gdb GUI frontends like `gdbgui`, or `gf` (https://github.com/nakst/gf) supports rr mode.

It will be definitely appealing to c/cpp community if you support rr with your debugger.

vimkim, 2024-07-20:
**Amazing news!** It just works with RR debugger!!

I had been trying to figure it out by myself for three days and almost got frustrated, but I eventually succeeded by thoroughly reading the RR debugger's documentation. It turned out to be quite simple.

# Steps to use gdb-frontend with RR debugger

### terminal 1

```sh
$ rr record <my program>
$ rr replay -s 9999 -k
...
```

### terminal 2

```sh
$ gdbfrontend
...
```

Then, go to the webpage and connect to localhost:9999

[image]

Then, type the following in the gdb command line

```sh
set sysroot /
```

[image]

Then set a breakpoint and continue and you are good to go. It works like a charm!!!!

The reverse commands, such as **reverse step**, **reverse continue**, and **reverse next**, are also displayed perfectly on the screen (unlike in VS Code, where they currently do not).

Thank you so much for creating such an amazing project. The only thing I would like to see added is support for reverse buttons, though they are not necessary since I can always use the command line.

I appreciate your hard work. If you don't mind, I recommend you to:

1. Specify in your `README.md` that your project works with `rr`.
2. Inform the `rr` community that your project is compatible with the `rr` debugger.

I believe many people like me would get to know about this project and appreciate your great work.

You are at the forefront of improving the welfare of Linux developers. Thanks again.