rohanrhu/gdb-frontend #69: Windows 10 x64 support?

g40, 2024-10-05:
Looks very interesting. If it can help with my MSYS64/gcc debugging problem then A+++.  Many thanks.

What have I missed here? What does `Tmux` do on Windows?

```
>python --version
Python 3.12.0

r:\src\qemu\qemu-w10-tap>pip install gdbfrontend
Collecting gdbfrontend
  Downloading gdbfrontend-0.11.3-py3-none-any.whl.metadata (18 kB)
Downloading gdbfrontend-0.11.3-py3-none-any.whl (17.8 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 17.8/17.8 MB 6.4 MB/s eta 0:00:00
Installing collected packages: gdbfrontend
Successfully installed gdbfrontend-0.11.3

r:\src\qemu\qemu-w10-tap>gdbfrontend
'gdbfrontend' is not recognized as an internal or external command,
operable program or batch file.
r:\src\qemu\qemu-w10-tap>finder -r r:\apps\python gdbfront*.*
r:\apps\python\Lib\site-packages\gdbfrontend\gdbfrontend.code-workspace
r:\apps\python\Lib\site-packages\gdbfrontend\gdbfrontend.py
r:\apps\python\Lib\site-packages\gdbfrontend\__pycache__\gdbfrontend.cpython-312.pyc
r:\apps\python\Lib\site-packages\gdbfrontend\__pycache__\gdbfrontend.cpython-39.pyc
r:\apps\python\Lib\site-packages\gdbfrontend\docs\media\gdbfrontend-icon.png
r:\apps\python\Lib\site-packages\gdbfrontend\docs\media\gdbfrontend-logo.png
r:\apps\python\Lib\site-packages\gdbfrontend\docs\media\gdbfrontend-screenshots.png
r:\apps\python\Lib\site-packages\gdbfrontend\documentation\src\media\gdbfrontend-icon.png
r:\apps\python\Lib\site-packages\gdbfrontend\documentation\src\media\gdbfrontend-logo.png
r:\apps\python\Lib\site-packages\gdbfrontend\documentation\src\media\gdbfrontend-screenshots.png
r:\apps\python\Lib\site-packages\gdbfrontend\frontend\components\GDBFrontend\css\GDBFrontend.css
r:\apps\python\Lib\site-packages\gdbfrontend\frontend\components\GDBFrontend\js\GDBFrontend.js
r:\apps\python\Lib\site-packages\gdbfrontend\media\gdbfrontend-screenshots.png
r:\apps\python\Lib\site-packages\gdbfrontend\media\gdbfrontend.png

r:\src\qemu\qemu-w10-tap>python -m gdbfrontend.py
R:\apps\python\python.exe: No module named gdbfrontend.py

r:\src\qemu\qemu-w10-tap>python -m gdbfrontend
[Error] Tmux is not installed. Please install Tmux on your system and run GDBFrontend again.

r:\src\qemu\qemu-w10-tap>pip install tmux
ERROR: Could not find a version that satisfies the requirement tmux (from versions: none)
ERROR: No matching distribution found for tmux

r:\src\qemu\qemu-w10-tap>pip install Tmux
ERROR: Could not find a version that satisfies the requirement Tmux (from versions: none)
ERROR: No matching distribution found for Tmux
```

rohanrhu, 2024-10-05:
GDBFrontend doesn't support Windows but you can use it on WSL (Windows Subsystem for Linux).

g40, 2024-10-05:
@rohanrhu Thank you. 

Sadly I cannot use WSL as I am building QEMU using MSYS64 so it generates Windows binaries. It is these I would like to debug. QEMU will not build on WSL.

What is it that prevents operation on Windows? What is missing?

rohanrhu, 2024-10-06:
> What is it that prevents operation on Windows? What is missing?

GDB doesn't support background execution on Windows.

You can run QEMU on WSL but I think it might not be able to use KVM on WSL. Another option, you can run GDBFrontend on WSL and connect to the Windows GDB that runs QEMU from WSL one.

g40, 2024-10-06:
Can you confirm: you need GDB to run as a detached process but ideally you would know its PID so it can be closed/killed? 
Does it use GDB-MI via a socket or some other IPC? 
Let me know as I think I may have a solution to your problem on Windows.

rohanrhu, 2024-10-06:
https://sourceware.org/gdb/wiki/GDB%20Front%20Ends

As you can see here, GDBFrontend doesn't use MI interface.

GDBFrontend runs on a thread in GDB process. All of GF APIs use GDB APIs that must be called on GDB's main thread but since background execution is unavailable on Windows, your debugged process is blocking GDB's main thread; GF pushes callbacks to run them on main thread but when main thread is blocked by the debugged process while running, it is not possible to get them called when they must be.